robgonnella/ardi #3: Feature request: Library dependency management

Pimmetje, 2019-09-23:
I would love to see something like composer is for PHP but than for Arduino.

Must have. 
* Be able to install a specific version or tag/branch for repeated builds. (composer lock)
* The option to use custom git repo's

Nice to have
* composer update like function (based on rules get the latest version)
* Auto search arduino default libraries like the IDE does. (without giving full repo paths). 

This would make the tool very handy for making a CI setup.
Currently i have all my libraries in one big repo. But it's getting way to big to handle. (Not in size but in numbers). And i can only build something with the latest version of my libraries. Updating external libraries is hard/tine consuming. This feature would be able to solve that.

Thanks for taking the time for this. I do not know the language GO. But if i can help with anything (like researching how the arduino ide works let me know.

robgonnella, 2019-09-24:
Thank you for your suggestion, this is a great idea! I've started working on a patch to add this feature and will keep you updated on my progress. It may take me some time before it's ready to publish.

Pimmetje, 2019-09-24:
If there is anything i can do please let me know.

Pimmetje, 2019-09-24:
Just fpr reference a description (including a download link for the official sources) can be found in there wiki https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

robgonnella, 2019-09-30:
It's still a ways off from being a complete feature but the basics are there. Still lots of work to do around error checking, validations, testing, documenting etc.

Feel free to give it a try in its primordial state and let me know what you think though! You can clone the repo and from the repo directory run

```
git fetch origin
git checkout add-library-manager
go install
```

From there you should have `ardi lib` and `ardi compile` commands

robgonnella, 2019-10-04:
The beginnings of this feature have been added
- [x] Be able to install a specific version for repeated builds. 
- [x] Search arduino default libraries

Still to come:
- [ ] The option to use custom git repo's
- [ ] Update function (based on rules get the latest version)

Pimmetje, 2019-10-17:
Will something like this help https://godoc.org/github.com/constabulary/gb/cmd/gb-vendor for the custom dependencies?

robgonnella, 2019-10-19:
Sorry for the delay, things have been busy past couple of weeks so I haven't had much time to work on this project. Should be able to get some more work in on this soon though!

Thanks for the link! I believe that library is only concerned with managing golang dependencies but I will look into that a bit to be sure. If not that one, there may be other golang tools out there to help manage remote dependencies.

Pimmetje, 2019-10-19:
Another one i found is https://github.com/golang/dep / https://golang.github.io/dep/docs/introduction.html

robgonnella, 2020-07-29:
There have been a lot of updates in v2, check it out and let me know what you think

closed 2020-12-04 by robgonnella