robgonnella/ardi #5: [master] Unable to init / ardi not placed in PATH

Pimmetje, 2019-09-30:
I installed ardi on a ubuntu 16.04 server. When i do 
/root/go/bin/ardi init
```
INFO[0000] Initializing. This may take some time...
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x53a1fb]

goroutine 22 [running]:
github.com/arduino/go-paths-helper.(*Path).Join(0x0, 0xc0000ad858, 0x1, 0x1, 0xc000236330)
        /root/go/src/github.com/arduino/go-paths-helper/paths.go:83 +0x4b
github.com/arduino/arduino-cli/configs.(*Configuration).LibrariesDir(...)
        /root/go/src/github.com/arduino/arduino-cli/configs/configuration.go:94
github.com/arduino/arduino-cli/commands.createInstance(0xd52ea0, 0xc0001ab530, 0xc0001507e0, 0x0, 0x0, 0xc0000ada00, 0x8ecc4f, 0x127f738, 0xc00003a960, 0x22, ...)
        /root/go/src/github.com/arduino/arduino-cli/commands/instances.go:314 +0x11b
github.com/arduino/arduino-cli/commands.Init(0xd52ea0, 0xc0001ab530, 0xc0001ab560, 0xc0000adb28, 0xc0000adb10, 0xc0001aa660, 0x7fc19f62c008, 0x0, 0xc0001ab560)
        /root/go/src/github.com/arduino/arduino-cli/commands/instances.go:150 +0x314
github.com/arduino/arduino-cli/commands/daemon.(*ArduinoCoreServerImpl).Init(0xc00000fe00, 0xc0001ab560, 0xd577a0, 0xc000236160, 0xc00000fe00, 0x20)
        /root/go/src/github.com/arduino/arduino-cli/commands/daemon/daemon.go:113 +0xf6
github.com/arduino/arduino-cli/rpc/commands._ArduinoCore_Init_Handler(0xc22880, 0xc00000fe00, 0xd564e0, 0xc0000f8180, 0x127f738, 0xc0000fbf00)
        /root/go/src/github.com/arduino/arduino-cli/rpc/commands/commands.pb.go:1394 +0x109
google.golang.org/grpc.(*Server).processStreamingRPC(0xc0000334a0, 0xd5ba60, 0xc0001a4d80, 0xc0000fbf00, 0xc0001aa6f0, 0x1254500, 0x0, 0x0, 0x0)
        /root/go/src/google.golang.org/grpc/server.go:1199 +0xb2e
google.golang.org/grpc.(*Server).handleStream(0xc0000334a0, 0xd5ba60, 0xc0001a4d80, 0xc0000fbf00, 0x0)
        /root/go/src/google.golang.org/grpc/server.go:1279 +0xd30
google.golang.org/grpc.(*Server).serveStreams.func1.1(0xc000035b30, 0xc0000334a0, 0xd5ba60, 0xc0001a4d80, 0xc0000fbf00)
        /root/go/src/google.golang.org/grpc/server.go:710 +0xbb
created by google.golang.org/grpc.(*Server).serveStreams.func1
        /root/go/src/google.golang.org/grpc/server.go:708 +0xa1
```

robgonnella, 2019-09-30:
Ahh yes sorry about this! I need to update the documentation for how to install. To install with go get you need to add `GO111MODULE=on` to the beginning. So the full command should be
```
GO111MODULE=on go get github.com/robgonnella/ardi
```

I will update the documentation tonight. Thank you for bringing this to my attention

Pimmetje, 2019-09-30:
The output is a bit longer than i would expect so ill paste it here (still need to test the rest)
```
/root/go/bin/ardi init
INFO[0000] Initializing. This may take some time...
instance:<id:1 > platforms_index_errors:"loading json index file /root/.ardi/arduino-rpc-client/package_index.json: open /root/.ardi/arduino-rpc-client/package_index.json: no such file or directory" libraries_index_error:"reading library_index.json: open /root/.ardi/arduino-rpc-client/library_index.json: no such file or directory"
.....................................................................................................................................................................................................................................................................................................................................................................................fatal error: runtime: out of memory

runtime stack:
runtime.throw(0xc6d19a, 0x16)
        /usr/lib/go-1.13/src/runtime/panic.go:774 +0x72
runtime.sysMap(0xc080000000, 0x8000000, 0x12afa78)
        /usr/lib/go-1.13/src/runtime/mem_linux.go:169 +0xc5
runtime.(*mheap).sysAlloc(0x12969c0, 0x8000000, 0x3, 0x3be00)
        /usr/lib/go-1.13/src/runtime/malloc.go:701 +0x1cd
runtime.(*mheap).grow(0x12969c0, 0x4000, 0x7f86ffffffff)
        /usr/lib/go-1.13/src/runtime/mheap.go:1252 +0x42
runtime.(*mheap).allocSpanLocked(0x12969c0, 0x4000, 0x12afa88, 0x41f7a5)
        /usr/lib/go-1.13/src/runtime/mheap.go:1163 +0x291
runtime.(*mheap).alloc_m(0x12969c0, 0x4000, 0x7f86f4690101, 0x7f86f4691a28)
        /usr/lib/go-1.13/src/runtime/mheap.go:1015 +0xc2
runtime.(*mheap).alloc.func1()
        /usr/lib/go-1.13/src/runtime/mheap.go:1086 +0x4c
runtime.(*mheap).alloc(0x12969c0, 0x4000, 0x7f86f4010101, 0x7f86f4691a28)
        /usr/lib/go-1.13/src/runtime/mheap.go:1085 +0x8a
runtime.largeAlloc(0x7fffe00, 0x7f86dcfb0101, 0x7f86f4691a28)
        /usr/lib/go-1.13/src/runtime/malloc.go:1138 +0x97
runtime.mallocgc.func1()
        /usr/lib/go-1.13/src/runtime/malloc.go:1033 +0x46
runtime.systemstack(0x7f86dd7b6bb0)
        /usr/lib/go-1.13/src/runtime/asm_amd64.s:370 +0x66
runtime.mstart()
        /usr/lib/go-1.13/src/runtime/proc.go:1146

goroutine 91 [running]:
runtime.systemstack_switch()
        /usr/lib/go-1.13/src/runtime/asm_amd64.s:330 fp=0xc00d9f2820 sp=0xc00d9f2818 pc=0x45aad0
runtime.mallocgc(0x7fffe00, 0xb30e60, 0x1, 0x0)
        /usr/lib/go-1.13/src/runtime/malloc.go:1032 +0x895 fp=0xc00d9f28c0 sp=0xc00d9f2820 pc=0x40e115
runtime.makeslice(0xb30e60, 0x7fffe00, 0x7fffe00, 0xc07bee63d0)
        /usr/lib/go-1.13/src/runtime/slice.go:49 +0x6c fp=0xc00d9f28f0 sp=0xc00d9f28c0 pc=0x44542c
bytes.makeSlice(0x7fffe00, 0x0, 0x0, 0x0)
        /usr/lib/go-1.13/src/bytes/buffer.go:229 +0x77 fp=0xc00d9f2958 sp=0xc00d9f28f0 pc=0x4f7c47
bytes.(*Buffer).grow(0xc011201c50, 0x200, 0xefa30)
        /usr/lib/go-1.13/src/bytes/buffer.go:142 +0x15b fp=0xc00d9f29a8 sp=0xc00d9f2958 pc=0x4f758b
bytes.(*Buffer).ReadFrom(0xc011201c50, 0xd63c20, 0xc0579bc2a0, 0x7f86f474f9d0, 0xc011201c50, 0x1)
        /usr/lib/go-1.13/src/bytes/buffer.go:202 +0x4b fp=0xc00d9f2a18 sp=0xc00d9f29a8 pc=0x4f7a3b
io.copyBuffer(0xd63060, 0xc011201c50, 0xd63c20, 0xc0579bc2a0, 0x0, 0x0, 0x0, 0xbc6d40, 0xc0573ea001, 0xc011201c50)
        /usr/lib/go-1.13/src/io/io.go:388 +0x2ed fp=0xc00d9f2a90 sp=0xc00d9f2a18 pc=0x46d1ed
io.Copy(...)
        /usr/lib/go-1.13/src/io/io.go:364
github.com/codeclysm/extract.copyCancel(0xd75360, 0xc01cf840c0, 0xd63060, 0xc011201c50, 0xd62f80, 0xc0003d9b00, 0x0, 0x1, 0x0)
        /root/go/pkg/mod/github.com/codeclysm/extract@v2.2.0+incompatible/cancelable_reader.go:10 +0xaf fp=0xc00d9f2af0 sp=0xc00d9f2a90 pc=0x7f01ef
github.com/codeclysm/extract.(*Extractor).Tar(0xc00d9f2d90, 0xd75360, 0xc01cf840c0, 0xd64340, 0xc057515920, 0xc00040ad40, 0x34, 0x0, 0xc6869c, 0x11)
        /root/go/pkg/mod/github.com/codeclysm/extract@v2.2.0+incompatible/extractor.go:145 +0x37c fp=0xc00d9f2d20 sp=0xc00d9f2af0 pc=0x7f128c
github.com/codeclysm/extract.Tar(...)
        /root/go/pkg/mod/github.com/codeclysm/extract@v2.2.0+incompatible/extract.go:90
github.com/codeclysm/extract.(*Extractor).Bz2(0xc00d9f3048, 0xd75360, 0xc01cf840c0, 0xd64340, 0xc0573edae0, 0xc00040ad40, 0x34, 0x0, 0xc6a157, 0x13)
        /root/go/pkg/mod/github.com/codeclysm/extract@v2.2.0+incompatible/extractor.go:67 +0x1f4 fp=0xc00d9f2df0 sp=0xc00d9f2d20 pc=0x7f0bd4
github.com/codeclysm/extract.(*Extractor).Archive(0xc00d9f3048, 0xd75360, 0xc01cf840c0, 0xd64340, 0xc0573edae0, 0xc00040ad40, 0x34, 0x0, 0x10, 0xb91880)
        /root/go/pkg/mod/github.com/codeclysm/extract@v2.2.0+incompatible/extractor.go:48 +0x1d1 fp=0xc00d9f2ec8 sp=0xc00d9f2df0 pc=0x7f0701
github.com/codeclysm/extract.Archive(...)
        /root/go/pkg/mod/github.com/codeclysm/extract@v2.2.0+incompatible/extract.go:50
github.com/arduino/arduino-cli/arduino/resources.(*DownloadResource).Install(0xc000458a50, 0xc0002b8b10, 0xc0002b8b20, 0xc0579cf080, 0x0, 0x0)
        /root/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20190722075247-15fcf19915f4/arduino/resources/install.go:61 +0x4f0 fp=0xc00d9f3078 sp=0xc00d9f2ec8 pc=0x7f58a0
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).InstallTool(0xc000070bc0, 0xc000557b30, 0xb, 0xc058a23a00)
        /root/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20190722075247-15fcf19915f4/arduino/cores/packagemanager/install_uninstall.go:85 +0x163 fp=0xc00d9f3118 sp=0xc00d9f3078 pc=0x806593
[... cut ...]
net/http.(*persistConn).writeLoop(0xc0005f06c0)
        /usr/lib/go-1.13/src/net/http/transport.go:2196 +0x123
created by net/http.(*Transport).dialConn
        /usr/lib/go-1.13/src/net/http/transport.go:1567 +0xb23

goroutine 244 [select, 13 minutes]:
go.bug.st/cleanup.InterruptableContext.func1(0xd75360, 0xc00010c400, 0xc000418ce0)
        /root/go/pkg/mod/go.bug.st/cleanup@v1.0.0/context.go:21 +0x11f
created by go.bug.st/cleanup.InterruptableContext
        /root/go/pkg/mod/go.bug.st/cleanup@v1.0.0/context.go:18 +0x6f
```

robgonnella, 2019-09-30:
Hmm that seems wrong, definitely not the output you should be seeing and it shouldn't take so long. For me it completes in just a few minutes. I'm using separate goroutines to install platforms in parallel to help speed up the process, I may need to limit the number of goroutines. For now can you run with -v (`ardi init -v`) to see command debug logs? I wouldn't sit through the whole initialization process either, just want to see if we can get to where the first error occurs. Also if you could share the output of running `lscpu` it may help me debug. Another thing I can add to help it the ability to only init for a specified platform instead of downloading and installing all of them

robgonnella, 2019-09-30:
Sorry I just saw your edits. 60 min is still too long, I'm wondering if that's network related. I'll try to run some tests and verify. Either way it seems adding a feature to only initialize for a specified platform could be a desired feature and should help with this issue.

Pimmetje, 2019-10-01:
Ill try again, i missed your comment. I have a very fast connection. Maybe that is causing the issue. Ill run it with -v and report back later today.

Pimmetje, 2019-10-01:
The downloading was done in minutes. The unpacking takes a long time. I guess because of IO limits and too many tasks. How many tasks are executed? 2 would do in my case i guess.

Pimmetje, 2019-10-02:
I did a clean & init. Did not have any issues this time. Ill do some more testing in the weekend.

robgonnella, 2019-10-02:
Ok great! Thank you for testing this!

Either way I have a patch coming up soon that will both limit the number of concurrent download / unpack jobs and additionally gives you an option to init for only a specified platform and an option to list available platforms so you can find your desired platform easier.

robgonnella, 2019-10-04:
- Resolved dependency issues so `GO111MODULE=on` is no longer required when installing although still recommended.
- Limited number of concurrent platform downloads / installs to 2
- Added ability to initialize only for a specified platform

See:
https://github.com/robgonnella/ardi/commit/8421d48864ab15ec555a61484a1fbae2591b1abe
https://github.com/robgonnella/ardi/pull/4

Pimmetje, 2019-10-04:
Took a clean installed ubuntu 18.04 with 1GB of ram.

using more or less those commands
```
sudo apt install software-properties-common -y && sudo add-apt-repository ppa:longsleep/golang-backports -y && sudo apt-get update && sudo apt-get install golang-go -y
GO111MODULE=on go get github.com/robgonnella/ardi
export PATH=$PATH:$(go env GOPATH)/bin
ardi init --verbose
```
(sorry for the looong logfile

```:~/go/bin# ./ardi init --verbose
INFO[0000] Initializing. This may take some time...
DEBU[0000] Creating data directory if needed
DEBU[0000] Starting daemon
DEBU[0000] TASK: name:"Downloading missing tool builtin:ctags@5.8-arduino11"
DEBU[0000] DOWNLOAD: url:"https://downloads.arduino.cc/tools/ctags-5.8-arduino11-pm-x86_64-pc-linux-gnu.tar.bz2" file:"builtin:ctags@5.8-arduino11" total_size:111604
DEBU[0000] DOWNLOAD: downloaded:111604
DEBU[0000] DOWNLOAD: completed:true
DEBU[0000] TASK: completed:true
DEBU[0000] TASK: name:"Installing builtin:ctags@5.8-arduino11"
DEBU[0000] TASK: message:"builtin:ctags@5.8-arduino11 installed" completed:true
DEBU[0000] TASK: name:"Downloading missing tool builtin:serial-discovery@0.5.0"
DEBU[0001] DOWNLOAD: url:"https://downloads.arduino.cc/tools/serial-discovery-linux64-v1.0.0.tar.bz2" file:"builtin:serial-discovery@0.5.0" total_size:1503971
DEBU[0001] DOWNLOAD: downloaded:645363
DEBU[0001] DOWNLOAD: downloaded:1503971
DEBU[0001] DOWNLOAD: completed:true
DEBU[0001] TASK: completed:true
DEBU[0001] TASK: name:"Installing builtin:serial-discovery@0.5.0"
instance:<id:1 > platforms_index_errors:"loading json index file /root/.ardi/arduino-rpc-client/package_index.json: open /root/.ardi/arduino-rpc-client/package_index.json: no such file or directory" libraries_index_error:"reading library_index.json: open /root/.ardi/arduino-rpc-client/library_index.json: no such file or directory"
DEBU[0001] TASK: message:"builtin:serial-discovery@0.5.0 installed" completed:true
DEBU[0001] Got a new instance with ID: 1
DEBU[0001] Updating index...
DEBU[0001] DOWNLOAD: url:"https://downloads.arduino.cc/packages/package_index.json" file:"Updating index: package_index.json" total_size:276721
DEBU[0001] DOWNLOAD: downloaded:276721
DEBU[0001] DOWNLOAD: completed:true
DEBU[0001] Update index done
DEBU[0001] Search result: Arrow: Arrow:samd - 2.1.0
DEBU[0001] Search result: atmel-avr-xminis: atmel-avr-xminis:avr - 0.6.0
DEBU[0001] Search result: littleBits: littleBits:avr - 1.0.0
DEBU[0001] Search result: Microsoft: Microsoft:win10 - 1.1.2
DEBU[0001] Search result: emoro: emoro:avr - 3.2.2
DEBU[0001] Search result: industruino: industruino:samd - 1.0.1
DEBU[0001] Search result: arduino: arduino:samd - 1.8.3
DEBU[0001] Search result: arduino: arduino:samd_beta - 1.6.25
DEBU[0001] Search result: arduino: arduino:mbed - 1.1.2
DEBU[0001] Search result: arduino: arduino:nrf52 - 1.0.2
DEBU[0001] Search result: arduino: arduino:avr - 1.8.1
DEBU[0001] Search result: arduino: arduino:megaavr - 1.8.5
DEBU[0001] Search result: arduino: arduino:sam - 1.6.12
DEBU[0001] Search result: Intel: Intel:i686 - 1.6.7+1.0
DEBU[0001] Search result: Intel: Intel:arc32 - 2.0.4
DEBU[0001] Search result: Intel: Intel:i586 - 1.6.7+1.0
DEBU[0001] Installing platform: i586:1.6.7+1.0
DEBU[0001] Installing platform: samd:2.1.0
DEBU[0001] Installing platform: avr:0.6.0
DEBU[0001] Installing platform: avr:1.0.0
DEBU[0001] Installing platform: win10:1.1.2
DEBU[0001] Installing platform: avr:3.2.2
DEBU[0001] Installing platform: samd:1.0.1
DEBU[0001] Installing platform: samd:1.8.3
DEBU[0001] Installing platform: samd_beta:1.6.25
DEBU[0001] Installing platform: mbed:1.1.2
DEBU[0001] Installing platform: nrf52:1.0.2
DEBU[0001] Installing platform: avr:1.8.1
DEBU[0001] Installing platform: megaavr:1.8.5
DEBU[0001] Installing platform: sam:1.6.12
DEBU[0001] Installing platform: i686:1.6.7+1.0
DEBU[0001] Installing platform: arc32:2.0.4
DEBU[0001] TASK: name:"Downloading packages"
[... 3277 lines cut ...]
        /root/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20190722075247-15fcf19915f4/arduino/cores/packagemanager/install_uninstall.go:85 +0x163
github.com/arduino/arduino-cli/commands.InstallToolRelease(0xc000234080, 0xc000516060, 0xc0004a1b20, 0xc00019cff0, 0x0)
        /root/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20190722075247-15fcf19915f4/commands/bundled_tools.go:51 +0x301
github.com/arduino/arduino-cli/commands/core.installPlatform(0xc000234080, 0xc0001a83f0, 0xc0002b4e40, 0x3, 0x4, 0xc0005e7b08, 0xc0004a1b20, 0xc00019cff0, 0xc129c0, 0xc0005cb560)
        /root/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20190722075247-15fcf19915f4/commands/core/install.go:99 +0x53b
github.com/arduino/arduino-cli/commands/core.PlatformInstall(0xd75420, 0xc0005b5740, 0xc0005cb560, 0xc0005e7b08, 0xc0005e7b20, 0xc00019cff0, 0x894858244c894800, 0x0, 0x10)
        /root/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20190722075247-15fcf19915f4/commands/core/install.go:55 +0x290
github.com/arduino/arduino-cli/commands/daemon.(*ArduinoCoreServerImpl).PlatformInstall(0xc0001ac480, 0xc0005cb560, 0xd7b340, 0xc000184e60, 0xc0001ac480, 0x20)
        /root/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20190722075247-15fcf19915f4/commands/daemon/daemon.go:139 +0xdf
github.com/arduino/arduino-cli/rpc/commands._ArduinoCore_PlatformInstall_Handler(0xc3fce0, 0xc0001ac480, 0xd79c60, 0xc00058c900, 0x12ad2f8, 0xc000580000)
[... cut ...]
google.golang.org/grpc.(*Server).serveStreams.func1.1(0xc00003b8f0, 0xc000176780, 0xd7d620, 0xc000177080, 0xc000580000)
        /root/go/pkg/mod/google.golang.org/grpc@v1.21.1/server.go:717 +0xbb
created by google.golang.org/grpc.(*Server).serveStreams.func1
        /root/go/pkg/mod/google.golang.org/grpc@v1.21.1/server.go:715 +0xa1

goroutine 88 [runnable]:
bytes.makeSlice.func1()
        /usr/lib/go-1.13/src/bytes/buffer.go:224 +0x68
bytes.makeSlice(0x1e00, 0xc000542000, 0x1e00, 0x1e00)
        /usr/lib/go-1.13/src/bytes/buffer.go:229 +0x9

robgonnella, 2019-10-04:
Thank you for the output! Although it's long, it's helpful. I'll need to run a profiler to get a better understanding of where performance and memory improvements can be made. I'll try to do that over the next week or so.

It may be that, because these files are so large, it just too expensive for some systems to run any of the full initialization process in parallel. For now I would find the exact platform you need and just use `ardi init <platformID>`. 

Once again thank you for running through these tests as it's hard for me to test every environment in my free time. Your feedback has been invaluable!

robgonnella, 2019-10-05:
I've removed all goroutines from the initialization process for now as I think this is safer until I can gather more data on cpu & memory usage. It's slower now that everything run synchronously but hopefully more stable.

https://github.com/robgonnella/ardi/pull/10

Pimmetje, 2019-10-06:
Works for me now on a 1GB ubuntu server.

closed 2019-10-06 by Pimmetje