robgonnella/ardi #7: board list missing FQBN

Pimmetje, 2019-10-04:
The board list is missing the FQBN
```
./ardi board list mega
instance:<id:1 >
Board                                             Platform                    FQBN
Arduino Yún                                       arduino:avr
Arduino Uno                                       arduino:avr
Arduino Uno WiFi                                  arduino:avr
Arduino Diecimila                                 arduino:avr
Arduino Nano                                      arduino:avr
Arduino Mega                                      arduino:avr
Arduino MegaADK                                   arduino:avr
Arduino Leonardo                                  arduino:avr
Arduino Leonardo Ethernet                         arduino:avr
Arduino Micro                                     arduino:avr
Arduino Esplora                                   arduino:avr
Arduino Mini                                      arduino:avr
Arduino Ethernet                                  arduino:avr
Arduino Fio                                       arduino:avr
Arduino BT                                        arduino:avr
Arduino LilyPadUSB                                arduino:avr
Arduino Lilypad                                   arduino:avr
Arduino Pro                                       arduino:avr
Arduino ATMegaNG                                  arduino:avr
Arduino Robot Control                             arduino:avr
Arduino Robot Motor                               arduino:avr
Arduino Gemma                                     arduino:avr
Adafruit Circuit Playground                       arduino:avr
Arduino Yún Mini                                  arduino:avr
Arduino Industrial 101                            arduino:avr
Linino One                                        arduino:avr
Arduino Uno WiFi Rev2                             arduino:megaavr
Arduino Nano Every                                arduino:megaavr
atmega168pb-xmini                                 atmel-avr-xminis:avr
atmega328pb-xmini                                 atmel-avr-xminis:avr
atmega328p-xmini                                  atmel-avr-xminis:avr
EMoRo 2560. Board based on ATmega 2560 MCU        emoro:avr

```

robgonnella, 2019-10-04:
The FQBN won't be filled in until that platform is installed aka `ardi init <platformID>`

Pimmetje, 2019-10-05:
Maybe it's better to put that in the title somewhere
FQBN (if installed) 
or something similar

robgonnella, 2019-10-07:
I decided to add it to the README as I think adding it to the title of the command output is a bit too vague.

closed 2019-10-10 by Pimmetje