robgonnella/ardi #9: Some issues/wishes with compile

Pimmetje, 2019-10-04:
The compile commands works for me but it's a bit picky given it you can assumes the ino has the same name as the folder. 

This works but gives a strage name as hex file (..arduino.avr.mega.hex) starting with a dot
```
 ~/go/bin/ardi compile ./ -f arduino:avr:mega:cpu=atmega2560
```

When the ino is not given it could take the current folder name as <foldername>,ino so you can use run ardi compile -f FQBN and it runs

Aso it would be nice if we can do verbose on compile. So we can see whats going on. 

Maybe something for a backlog

Custom -D arguments for the compile (I use it for -DDEBUG so i can do a debug vs production build. 
Also we should consider saving the default values (FQBN/other arguments in the project file where also the library versions will be stored.

robgonnella, 2019-10-05:
Fixed issues with compile command and added the option to run verbosely. I removed the feature that allows you to define your sketches in a "sketches" directory as I don't think that's a real world use case. You can now run `ardi compile . -f arduino:avr:mega:cpu=atmega2560 -v` and everything should work as expected.
https://github.com/robgonnella/ardi/pull/10

That's a great idea about the custom -D arguments! I'll add that to the list. Thanks again!

Pimmetje, 2019-10-05:
I would make the . optional. As the command get a bit cleaner.

Thanks for your hard work & speed. I have had this kind of tool on my wish list for so long. But never got around building one. So when i found your project i was very happy :). Thanks again for your hard work!

robgonnella, 2019-10-05:
My pleasure! I'm happy you find this tool useful, that was my aim when I started building it! 😃 

I made the compile command default to current directory btw. So now you can run `ardi compile -f arduino:avr:mega:cpu=atmega2560 -v` and get the desired result :)

Pimmetje, 2019-10-05:
That works great thx. I made a bit of a mess putting multiple issues into one. 
Should i make a new one for the build arguments and the build definition in the config or will we keep this one open?

robgonnella, 2019-10-07:
Lets split the custom build arguments and build definition to a separate issue / feature request. It will be easier for me to track that way

Pimmetje, 2019-10-10:
Closing in favor of 14 and #15

closed 2019-10-10 by Pimmetje