robgonnella/ardi #11: list of supported boards not in same order

Pimmetje, 2019-10-05:
The list of supported boards is not in the same order. So for manually compiling the number changes. 

```
ardi compile .
instance:<id:1 >
No.        Name                                     FQBN
0          Arduino Nano                             arduino:avr:nano
1          Arduino Leonardo ETH                     arduino:avr:leonardoeth
2          LilyPad Arduino                          arduino:avr:lilypad
3          Arduino/Genuino Mega or Mega 2560        arduino:avr:mega
4          Adafruit Circuit Playground              arduino:avr:circuitplay32u4cat
5          Arduino/Genuino Uno                      arduino:avr:uno
6          Arduino Mega ADK                         arduino:avr:megaADK
7          LilyPad Arduino USB                      arduino:avr:LilyPadUSB
8          Arduino NG or older                      arduino:avr:atmegang
9          Arduino Pro or Pro Mini                  arduino:avr:pro
10         Arduino Mini                             arduino:avr:mini
11         Arduino BT                               arduino:avr:bt
12         Arduino Robot Motor                      arduino:avr:robotMotor
13         Arduino/Genuino Micro                    arduino:avr:micro
14         Arduino Ethernet                         arduino:avr:ethernet
15         Arduino Leonardo                         arduino:avr:leonardo
16         Arduino Gemma                            arduino:avr:gemma
17         Arduino Industrial 101                   arduino:avr:chiwawa
18         Arduino Uno WiFi                         arduino:avr:unowifi
19         Arduino Robot Control                    arduino:avr:robotControl
20         Arduino Yún Mini                         arduino:avr:yunmini
21         Linino One                               arduino:avr:one
22         Arduino Yún                              arduino:avr:yun
23         Arduino Duemilanove or Diecimila         arduino:avr:diecimila
24         Arduino Fio                              arduino:avr:fio
25         Arduino Esplora                          arduino:avr:esplora
26         Arduino Uno WiFi Rev2                    arduino:megaavr:uno2018
27         Arduino Nano Every                       arduino:megaavr:nona4809
```

```
ardi compile . --verbose
No.        Name                                     FQBN
0          Arduino/Genuino Uno                      arduino:avr:uno
1          Arduino Yún Mini                         arduino:avr:yunmini
2          LilyPad Arduino USB                      arduino:avr:LilyPadUSB
3          Arduino Leonardo ETH                     arduino:avr:leonardoeth
4          Arduino Mini                             arduino:avr:mini
5          Arduino/Genuino Micro                    arduino:avr:micro
6          Arduino Robot Motor                      arduino:avr:robotMotor
7          Arduino Uno WiFi                         arduino:avr:unowifi
8          Arduino Ethernet                         arduino:avr:ethernet
9          Arduino Industrial 101                   arduino:avr:chiwawa
10         Arduino Fio                              arduino:avr:fio
11         Arduino Gemma                            arduino:avr:gemma
12         Arduino Leonardo                         arduino:avr:leonardo
13         Arduino Esplora                          arduino:avr:esplora
14         Arduino NG or older                      arduino:avr:atmegang
15         Arduino/Genuino Mega or Mega 2560        arduino:avr:mega
16         Linino One                               arduino:avr:one
17         Arduino Mega ADK                         arduino:avr:megaADK
18         Arduino Nano                             arduino:avr:nano
19         Arduino Robot Control                    arduino:avr:robotControl
20         Arduino Yún                              arduino:avr:yun
21         Arduino BT                               arduino:avr:bt
22         Arduino Duemilanove or Diecimila         arduino:avr:diecimila
23         Adafruit Circuit Playground              arduino:avr:circuitplay32u4cat
24         Arduino Pro or Pro Mini                  arduino:avr:pro
25         LilyPad Arduino                          arduino:avr:lilypad
26         Arduino Uno WiFi Rev2                    arduino:megaavr:uno2018
27         Arduino Nano Every                       arduino:megaavr:nona4809
```

robgonnella, 2019-10-05:
Fixed! Output is now alphabetized
https://github.com/robgonnella/ardi/pull/13

Pimmetje, 2019-10-05:
Seems to do it thx

closed 2019-10-05 by Pimmetje