robgonnella/ardi #12: Question about the used compile tool version

Pimmetje, 2019-10-05:
When u use the arduino IDE u have version like 0.x 1.x 1.6.x 1.8.x. What version is used when u do a ardi init and is it possible to define this version?

I have had projects that did not work on specific versions of the arduino build tools. So this is maybe also something to consider. 
I do not think it's needed to have several versions installed at the same time. But it would be a nice to have :sweat_smile:

robgonnella, 2019-10-05:
Updated so you can now initialize for a specific version via `ardi init <platform_id@version>`.
https://github.com/robgonnella/ardi/pull/13

Unfortunately, at this time I cannot provide a list of available versions for each platform due to limitations in arduino-cli. I'll see if I can submit a patch upstream to add that ability, or maybe just submit a feature request via their issues page.

robgonnella, 2019-10-05:
Oh also, arduino-cli does not support wild card minor and patch versions so you'll have to supply the exact semantic version for now.

Pimmetje, 2019-10-05:
Nice, thx. It's a nice to have to install wildcards. I think in most cases u know the full version number except for latest maybe.

closed 2019-10-05 by Pimmetje