robgonnella/ardi #17: Ardi deployement error

klorydryk, 2020-01-25:
Hello,
I get an error using "go get github.com/robgonnella/ardi" :
`package github.com/robgonnella/ardi/v3/commands: cannot find package "github.com/robgonnella/ardi/v3/commands" in any of:
	/usr/lib/go-1.7/src/github.com/robgonnella/ardi/v3/commands (from $GOROOT)
	/home/pi/go/src/github.com/robgonnella/ardi/v3/commands (from $GOPATH)
`

robgonnella, 2020-01-26:
Can you try
```
GO111MODULE=on go get github.com/robgonnella/ardi@latest
```

klorydryk, 2020-01-27:
```
$GO111MODULE=on go get github.com/robgonnella/ardi@latest
package github.com/robgonnella/ardi@latest: invalid github.com/ import path "github.com/robgonnella/ardi@latest"
```

robgonnella, 2020-01-27:
Can you share a bit about your environment? What version of go are you on? Are you running this command from inside our outside of a go module?

robgonnella, 2020-01-27:
Installing with `go get` is only supported for go versions >= 1.12. However I just published the binary to github releases so you can also install with the following commands

```bash
curl -L https://github.com/robgonnella/ardi/releases/latest/download/ardi --output /usr/local/bin/ardi
chmod +x /usr/local/bin/ardi
```

klorydryk, 2020-01-27:
As I said, I use go version go1.7.4 linux/arm on a raspberry v1
Distrib is Debian 9.11

I did as you said, but the first command give me:
`-bash: /usr/local/bin/ardi : impossible d'exécuter le fichier binaire : Erreur de format pour exec()`
and with sudo
`/usr/local/bin/ardi: 3: /usr/local/bin/ardi: Syntax error: ")" unexpected`

robgonnella, 2020-01-27:
Apologies. I uploaded versions for ARM but I don't have a way to test this at the moment. You can try it out though using the below commands. Let me know how it goes. I am going to try to get set up to test this more soon.
```
curl -L https://github.com/robgonnella/ardi/releases/latest/download/ardi_linux_arm6 --output /usr/local/bin/ardi
chmod +x /usr/local/bin/ardi
```

robgonnella, 2020-02-20:
Were you able to use one of the ARM binaries? You can find all the prebuilt binaries here: https://github.com/robgonnella/ardi/releases

closed 2020-07-29 by robgonnella