robgonnella/ardi #51: CONSIDER wrapping console output to 80 characters

egnor, 2021-07-02:
(Thanks for ardi! It is exactly what I was planning to write, but already written! 🙏)

This is minor and please feel free to ignore (not meaning to start a holy war), but on my regular xterms, ardi's output looks like this:

```
% ardi attach-and-watch --help

Compile, upload, watch board logs, and watch for sketch changes. Updates to the 
sketch file will trigger automatic recompile, reupload, and restarts the board l
og watcher. If the sketch argument matches a user defined build in ardi.json, th
e build values will be used for compilation, upload, and watch path

...
```

... which is a bit unattractive and hard to read, compared to word-wrapped text. Maybe you run everything in some terminal environment that always word-wraps?

robgonnella, 2021-07-03:
Thanks for your suggestion, and thanks for trying out ardi! You're right this would be a good thing to get right. I'll add it to my list :)

robgonnella, 2021-07-18:
Updated in new beta version, you can try it out via `GO111MODULE=on go get github.com/robgonnella/ardi/v2@v2.3.0-beta.1`

egnor, 2021-07-30:
Looks good!! :+1:

robgonnella, 2021-08-03:
Closing: Addressed in v2.3.0

commit aaca1012 (2021-07-04): Wrap help text

Adds custom help function to wrap help text to 80 chars, or to the
terminal width if it's less than 80 chars.

closes #51
closed 2021-08-03 by robgonnella