robgonnella/ardi #52: CONSIDER making the "board log watcher" available outside of attach-and-watch

egnor, 2021-07-02:
(Thanks for ardi! It is exactly what I was planning to write, but already written! 🙏)

The "board log watcher" (a trivial arduino-to-host-only serial terminal) is great and exactly ideal for many cases. (More complicated cases, e.g. where input is desired, can use an actual terminal program.)

However at present it's only available in the "attach-and-watch" command, which staples it to various other quite helpful but not always exactly needed behavior. It would be nice to be able to say "ardi output" or something to grab whatever the board is outputting.

This is a low priority nice-to-have, since one can always go use another serial terminal program.

robgonnella, 2021-07-03:
Another great suggestion and pretty easy to knock out too! All the hookups are already there. I'll get going on this one soon!

robgonnella, 2021-07-18:
Updated in new beta version, you can try it out via `GO111MODULE=on go get github.com/robgonnella/ardi/v2@v2.3.0-beta.1`

egnor, 2021-07-30:
Works for me! A couple caveats:

- It defaults to some baud rate like 9600? Easily overridden with `-b 115200` or whatnot, but probably better to just use whatever baud rate the port is set on if not specified, and also print what it's doing?
- Reopening the port tends to reset the Arduino (Uno in this case) based on DTR toggling (I think). Not sure if there's any way to avoid this?

robgonnella, 2021-08-02:
I haven't found any way of reliably detecting what baud rate the port is set on other than trying to parse it from sketch files. In the standalone attach feature there is no sketch to parse so I wouldn't even be able to guess.

I've noticed that attaching to the port resets the mega 2560 too. Haven't found a solution yet so I've opened a separate issue for tracking: https://github.com/robgonnella/ardi/issues/85

robgonnella, 2021-08-03:
Closing and continuing investigation in issue #85

closed 2021-08-03 by robgonnella