robgonnella/ardi #53: CONSIDER allowing baud rate selection for board log monitoring in attach-and-watch

egnor, 2021-07-02:
(Thanks for ardi! It is exactly what I was planning to write, but already written! 🙏)

The little mini serial monitor in attach-and-watch is great, especially with how it integrates with the other attach-and-watch functionality (stopping during upload). But it seems to assume 115200 baud? That's a reasonable assumption to be sure but it seems worth adding some sort of configuration parameter (maybe something that can end up in ardi.json?).

This is low priority, since most sensible people use 115200 baud anyway, it's the fastest widely supported serial baud rate.

robgonnella, 2021-07-03:
Ahh so in this case ardi is just being somewhat smart, it's reading your sketch file and finding any defined baud rates, if it doesn't find one it actually defaults to `9600`. That being said, the logic around this isn't rock solid as there are definitely cases it would miss. I don't see a reason not to allow the user to define specific bauds for their builds. I can see it being a more intuitive user experience that way :)

If you're interested in the relevant code it's here: https://github.com/robgonnella/ardi/blob/main/v2/util/util.go#L281

egnor, 2021-07-03:
Ooh, tricksy!

egnor, 2021-07-03:
I guess I might propose this?
- if baud rate specified in build, use that (and maybe print that it's being used?)
- otherwise, if baud setting found by grepping code, use that, and print a notice to that effect
- otherwise, give an error (rather than defaulting to 9600) asking for it to be explicitly set

That would only go wrong for a false positive code-grep (say, a commented out baud setting), and even in that case the printed notice should give enough of a bread-crumb to give the user a fighting chance to figure out what's wrong (and set baud rate explicitly to whatever they want).

robgonnella, 2021-07-03:
Sounds reasonable to me!

robgonnella, 2021-07-18:
Updated in new beta version, you can try it out via `GO111MODULE=on go get github.com/robgonnella/ardi/v2@v2.3.0-beta.1`

egnor, 2021-07-30:
Verified works for me! The only caveat is I don't see a log message in the `attach-and-watch mysketch.ino` case when it's detecting baud rate.

robgonnella, 2021-08-03:
Closing: Addressed in v2.3.0

closed 2021-08-03 by robgonnella