robgonnella/ardi #54: FQBN autodetection for attach-and-watch

egnor, 2021-07-02:
(Thanks for ardi! It is exactly what I was planning to write, but already written! 🙏)

If I say `ardi upload mysketch.ino`, and a single board is connected and can be detected, the FQBN is automatically set up:

```
% ardi upload cap1188_tester.ino          
ardi    Uploading...                                  build=/home/egnor/source/m
achine/arduino/TESTS/cap1188_tester device=/dev/ttyACM0 fqbn="arduino:avr:uno"
ardi    Upload successful                            
```

That's nifty and handy. However, `ardi attach-and-watch mysketch.ino` doesn't do this:

```
% ardi attach-and-watch cap1188_tester.ino                   
ardi    Compiling...                                  fqbn= sketch=/home/egnor/s
ource/machine/arduino/TESTS/cap1188_tester/cap1188_tester.ino
Error: no FQBN provided
Usage:
  ardi attach-and-watch [sketch|build] [flags]

Flags:
  -p, --build-prop stringArray   Specify build property to compiler
  -f, --fqbn string              Specify fully qualified board name
  -h, --help                     help for attach-and-watch
      --port string              The port your arduino board is connected to

Global Flags:
  -q, --quiet     Silence all logs
  -v, --verbose   Print all logs

ardi   Command failed                                error="no FQBN provided"
```

Be a lot cooler if it did! Especially since `attach-and-watch` is the sort of thing that gets used interactively in a quick way, so it would be nice to minimize what has to be selected to run it.

robgonnella, 2021-07-03:
Ooh nice catch! Yes this should definitely work the way you described. I'll get this working soon

robgonnella, 2021-07-18:
Updated in new beta version, you can try it out via `GO111MODULE=on go get github.com/robgonnella/ardi/v2@v2.3.0-beta.1`

egnor, 2021-07-30:
Verified works for me! Super awesome. :fireworks:

robgonnella, 2021-08-03:
Closing: Addressed in v2.3.0

closed 2021-08-03 by robgonnella