robgonnella/ardi #55: Sketch (.ino) file monitoring isn't working for me in attach-and-watch

egnor, 2021-07-02:
(Thanks for ardi! It is exactly what I was planning to write, but already written! 🙏)

As I understand it, `ardi attach-and-watch` is supposed to monitor the .ino file so if I edit and save, it will rebuild and reflash. However, this doesn't seem to work.

```
% ardi attach-and-watch -f arduino:avr:uno cap1188_tester.ino
ardi    Compiling...                                  fqbn="arduino:avr:uno" ske
tch=/home/egnor/source/machine/arduino/TESTS/cap1188_tester/cap1188_tester.ino
Sketch uses 5552 bytes (17%) of program storage space. Maximum is 32256 bytes.
Global variables use 532 bytes (25%) of dynamic memory, leaving 1516 bytes for l
ocal variables. Maximum is 2048 bytes.
ardi    Watching /home/egnor/source/machine/arduino/TESTS/cap1188
_tester/cap1188_tester.ino for changes 
ardi    Watching logs...                             
CAP1188 tester running
Product ID: 0x50
Manuf. ID: 0x5D
Revision: 0x83
CAP1188 init done
```

(The sketch continues running, and I still see output from it as it's generated.) Even though it says "Watching ... for changes", if I go save a change to cap1188_tester.ino in my editor, or even just `touch cap1188_tester.ino`, nothing happens. (Re-running the `ardi attach-and-watch` command does pick up the change, of course.)

My system is a relatively ordinary x86-64 desktop running Ubuntu 21.04. Further diagnostics available if it's helpful! I can also try `strace` and similar to try to see what's going on under the covers.

robgonnella, 2021-07-03:
Thanks for reporting this! I will investigate this more and roll out a fix soon

robgonnella, 2021-07-06:
I'm having trouble reproducing this although I did find a different bug with the timing of closing the serial connection before re-uploading. I'll be working on a patch for that next week. In the meantime do you think you could run the command with `-v` and post the logs?

```
ardi attach-and-watch -f arduino:avr:uno cap1188_tester.ino -v
```

egnor, 2021-07-06:
(Will do when I get a chance!)

egnor, 2021-07-07:
When I do, I see:

```
ardi    Watching mpr121_tester.ino for changes       
ardi    Watching logs...                             
=== MPR121 tester starting ===
ardi   event: "mpr121_tester.ino": RENAME           
ardi   event: "mpr121_tester.ino": CHMOD            
ardi   event: "mpr121_tester.ino": REMOVE           
```

The RENAME/CHMOD/REMOVE lines come when I save the file in my editor (Vim). However, ardi doesn't reload the file. I guess you're just looking at WRITE events or something like that, versus move-and-replace operations?

On subsequent writes I don't see further lines, presumably because it's watching the old file that has now been renamed and deleted? (I don't know the semantics of the file watching system you're using.)

robgonnella, 2021-07-09:
Odd this is working fine for me using Vim and ubuntu (18.04, 20.04, & 21.04). I see `WRITE` events when saving. I'm not sure it makes sense to trigger recompilation and reuploading on `RENAME`, `CHMOD`, and `REMOVE` events.

egnor, 2021-07-10:
The `fsnotify` docs are vague on semantics, but I _think_ you do have to look for `Create` and maybe `Rename` (but not `Chmod` or `Remove`) events in addition to `Write`?

I don't know why our versions of vim behave differently, but it's good editor save behavior to
1) write `foo.ino.tmp` with the new file
2) [optional] delete `foo.ino`
3) rename `foo.ino.tmp` to `foo.ino` (replacing the original if not already deleted)

That way the file gets updated atomically, so there's no chance of picking up a half-written file. However, when watching `foo.ino`, only a RENAME (and maybe REMOVE) event will be generated.

On Linux, the underlying `inotify` API provides more granular events (https://man7.org/linux/man-pages/man7/inotify.7.html). Using that API, we'd want to watch `IN_CLOSE_WRITE` and `IN_MOVED_TO` (but _not_ `IN_MODIFY`, to minimize the potential to read a half-written file). However, `fsnotify` (https://github.com/fsnotify/fsnotify/blob/master/inotify.go#L319) doesn't watch for `IN_CLOSE_WRITE` so we must accept the race. It seems to map `IN_MOVE_TO` to `Create`, which doesn't quite correspond with observed behavior on my system; I can dig deeper if desired.

robgonnella, 2021-07-11:
Thanks for investigating this more and opening up an issue with `fsnotify`! After a bit of playing around with this locally here is what I've discovered on ubuntu

```
cp file.tmp file --> WRITE
mv file file.tmp --> RENAME
mv file.tmp file --> CHMOD, REMOVE
```

Based on this I've created a PR that watches for the `REMOVE` event as it seems as though that is the last event in an atomic update, and then checks if the target file still exists, if so it re-adds it to the watcher and continues with the update, if not we bail.

robgonnella, 2021-07-18:
Updated in new beta version, you can try it out via `GO111MODULE=on go get github.com/robgonnella/ardi/v2@v2.3.0-beta.1`

egnor, 2021-07-30:
Verified works for me! :tada:

robgonnella, 2021-08-03:
Closing: Addressed in v2.3.0

closed 2021-08-03 by robgonnella