robgonnella/ardi #56: CONSIDER defaults for sketch/build name

egnor, 2021-07-02:
(Thanks for ardi! It is exactly what I was planning to write, but already written! pray)

For quick use, it would be great to just say `ardi attach-and-watch` or `ardi build` or whatever, without having to specify a build or sketch name.

As a proposal:
- If only one build is defined, use it
- If a build named "default" is defined, use it
- Otherwise, if multiple builds are defined, error (require picking one)
- If no builds are defined, but there's a sketch matching the current directory name, use it
- Otherwise, error (require picking something)

Of course a default sketch would still require FQBN selection, but for `attach-and-watch` or `upload`, FQBN can be autodetected.

There's some risk here of being "too magical", but I think the proposal above is on the good side of that line. Your opinion is what matters though!

robgonnella, 2021-07-03:
I like it! Shouldn't be too hard to implement, again a lot of the hook-ups are already there. Thanks for taking the time to come up with these great suggestions! I'll get started on these enhancements soon

robgonnella, 2021-07-18:
Updated in new beta version, you can try it out via `GO111MODULE=on go get github.com/robgonnella/ardi/v2@v2.3.0-beta.1`

robgonnella, 2021-08-03:
Closing: Addressed in v2.3.0

closed 2021-08-03 by robgonnella