robgonnella/ardi #60: Default ardi.json includes absolute paths, which is problematic when checked in

egnor, 2021-07-06:
`ardi add build` adds entries like this by default to `ardi.json`:

```
	"builds": {
		"mpr121_mega": {
			"directory": "/home/egnor/source/machine/arduino/TESTS/mpr121_tester",
			"sketch": "/home/egnor/source/machine/arduino/TESTS/mpr121_tester/mpr121_tester.ino",
			"baud": 115200,
			"fqbn": "arduino:avr:mega",
			"props": {}
		}
	}
```

Since an `ardi.json` file is often checked in to source control (e.g. git) and checked out elsewhere, this is problematic since the absolute paths don't work elsewhere.

For me, at least, changing "directory" to "." and "sketch" to a local path seems to work, though this may have other side effects?

robgonnella, 2021-07-10:
good catch, will update this to use relative paths in `ardi.json` and construct absolute paths for build and upload requests

robgonnella, 2021-07-18:
Updated in new beta version, you can try it out via `GO111MODULE=on go get github.com/robgonnella/ardi/v2@v2.3.0-beta.1`

egnor, 2021-07-30:
Verified works!

robgonnella, 2021-08-03:
Closing: Addressed in v2.3.0

closed 2021-08-03 by robgonnella