robgonnella/ardi #61: CONSIDER a central (per-user) cache of large items

egnor, 2021-07-06:
The `.ardi` directory in a project is typically quite large, e.g. 295M for a very simple project I just set up. In some cases (such as mine) I may end up with a number of different projects set up on the same machine. It would be nice if common large items could be linked to a central cache directory. (Clearly we'd want to make sure this is only done if they are exactly the same item.)

There are tradeoffs here to be sure (who cleans the cache?), but it is a popular approach taken by some other dependency managers (but not all of them).

I'm feeling the extra pain because I happen to be using this on a Raspberry Pi (among other places)... so I'm having to blow away `.ardi` directories occasionally to keep the disk from filling.

robgonnella, 2021-07-10:
This is a bit tricky, and possibly impossible at the moment, as I believe the underlying tool `arduino-cli` only supports using one data directory at a time. So your build can only utilize either a global data directory or a local project data directory but not both simultaneously. I'll do some digging around in `arduino-cli` to see if they've added support for multiple data directories recently or see if I can maybe submit a patch upstream if it's not too difficult, or even just open an issue.

egnor, 2021-07-10:
You could do something with symlinks maybe, but that could be fragile and
platform specific...

robgonnella, 2021-07-11:
A bit more investigation into this one shows that `arduino-cli` doesn't support having multiple versions of platforms installed simultaneously :-/

``` bash
ardi add platform arduino:avr
# results in .ardi/packages/arduino/hardware/avr/1.8.3
ardi add platform arduino:avr@1.8.2
# results in replacement of 1.8.3 directory with 1.8.2
```