robgonnella/ardi #62: CONSIDER having attach-and-watch install platforms if needed

egnor, 2021-07-06:
Running `ardi attach-and-watch` seems to install libraries if needed, but if a platform isn't installed, there's an error:

```
$ ~/go/bin/ardi attach-and-watch cap1188_mega
ardi    Compiling...                                  fqbn="arduino:avr:mega" sketch=cap1188_tester.ino
Downloading missing tool builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 downloaded                                                                                       
Installing builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:serial-discovery@1.2.1...
builtin:serial-discovery@1.2.1 downloaded                                                                                    
Installing builtin:serial-discovery@1.2.1...
builtin:serial-discovery@1.2.1 installed
Updating index: library_index.json downloaded                                                                                
Updating index: package_index.json downloaded                                                                                
Updating index: package_index.json.sig downloaded                                                                            
Error: platform not installed
Usage:
  ardi attach-and-watch [sketch|build] [flags]

Flags:
  -p, --build-prop stringArray   Specify build property to compiler
  -f, --fqbn string              Specify fully qualified board name
  -h, --help                     help for attach-and-watch
      --port string              The port your arduino board is connected to

Global Flags:
  -q, --quiet     Silence all logs
  -v, --verbose   Print all logs

ardi   Command failed                                error="platform not installed"
```

Of course this is trivially remedied with a run of `ardi install` but it would be nifty for `attach-and-watch` to just Do The Sensible Thing on a machine that hadn't previously been install-ed.

(Of course, it should avoid pinging the network if things _are_ installed, because it might be used offline.)

robgonnella, 2021-07-11:
In this case what you are seeing being auto-installed are the just the internal tools `arduino-cli` uses for auto-discovery, and the index files for platforms and libraries. This is out of the control of `ardi` as it's managed by `arduino-cli`; however, I've noticed that `arduino-cli` does not re-download the tools if they already exist in the data directory. As for user defined project platforms and libraries, I'd rather keep it simple for now and let the user run `install` when needed. I'll keep this issue open though as I may revisit this later.

egnor, 2021-07-11:
That makes sense, and it's no big deal for sure. The status quo does have
some arguments in favor:

The status quo does mirror other similar systems. (Some do auto-install,
like npx, but that's a bit of an outlier; "npm run" does require "npm
install" for example.)

The status quo keeps the operation local only (at least as far as ardi is
concerned) so when you're flashing arduinos on an airplane flight*, it
doesn't surprise you by trying to download something. (Though I gather
arduino-cli will assume it has network available as needed?)

* I may or may not have done this. Recommended only when everyone in your
line of sight knows you!

robgonnella, 2021-07-12:
Ha!! I bet that got a look or two from flight attendants! ✈️ 🤖 😆