robgonnella/ardi #68: WatchSketch is deadlock susceptible (uses fsnotify without separate goroutine)

egnor, 2021-07-10:
According to https://github.com/howeyc/fsnotify/issues/7, https://github.com/fsnotify/fsnotify/issues/55 and https://github.com/fsnotify/fsnotify#faq, callers need to start a separate goroutine to select on watcher events (vs where `fsnotify.NewWatcher()` is called). However, currently `WatchSketch()` is selecting events in the same thread where the watcher is created, which (apparently) may deadlock.

Disclaimer: I am not very go-literate, so, could easily be misunderstanding things somehow.

robgonnella, 2021-07-11:
Nice find! Just pushed up changes to address this :)

robgonnella, 2021-07-18:
Updated in new beta version, you can try it out via `GO111MODULE=on go get github.com/robgonnella/ardi/v2@v2.3.0-beta.1`

egnor, 2021-07-30:
Hard to verify but the fix certainly looks good :slightly_smiling_face:

robgonnella, 2021-08-03:
Closing: Addressed in v2.3.0

closed 2021-08-03 by robgonnella