robgonnella/ardi #85: Attaching to board triggers reset

robgonnella, 2021-08-02:
When using the standalone "attach" command, the board is reset every time. This is problematic if you wish to just attach and view logs at a given point in time.

egnor, 2021-09-05:
Something about the serial port close/open cycle toggling DTR, at least on some platforms?

robgonnella, 2021-09-06:
I've looked into this a bunch now and yes it is due to sending DTR signal when opening the serial port. This is not something that can really be changed in `ardi` though as I believe this is controlled via OS. In linux I think you need to actually modify the kernel, not sure what would be needed on OSX. Either way, if you did disable this signal, you would break the uploading feature as it relies on DTR and RTS signals to reset the board before uploading. You would have to manually press the reset button on the board at the exact right time when uploading. From what I've read the best option is to temporarily disable it via hardware approach. There are a number of posts about this, here is one: https://forum.arduino.cc/t/how-to-arduino-serial-communication-open-without-restart/536817/13

```
The arduino resets when you open a serial connection to it because of the "auto-reset" feature.
There are several ways to disable auto-reset:

• Connect a relatively large (10uF) capacitor between the RESET and GROUND on the power header of the board.
• Cut the RESET-EN solder jumper intended for this purpose (not all clones will have this jumper)
• remove the auto-reset capacitor in between DTR (from the USB/Serial chip) and the RESET signal.

All of these will interfere with the automatic upload of new sketches; you'll need to either undo them, or manage to use the manual reset button at the right time during upload (you have about a 2s window)
```

It seems another option is to connect to the Arduino using a USB-TTL cable that is connected to Rx Tx and GND.

For what it's worth I did try disabling DTR via software approach (golang serial package), but it can only be toggled after the unix socket is created which is too late and would really only be used if you wanted to manually trigger a board reset while already connected to the serial port.

egnor, 2021-09-06:
For Linux, at least, I'm seeing some conflicting info.

- https://unix.stackexchange.com/questions/446088/how-to-prevent-dtr-on-open-for-cdc-acm claims that the only way to avoid DTR toggle on port open/close is by kernel modification, _despite the setting of HUPCL_.

- But, https://raspberrypi.stackexchange.com/questions/9695/disable-dtr-on-ttyusb0/31298#31298 (which is widely cited, and despite the forum not Pi-specific) claims that clearing HUPCL on the port will prevent DTR toggle on future opens.

I can run some tests to try to get to whether clearing HUPCL does it fact avoid toggling DTR. On non-Linux platforms there may be other options. (Not sure what your host platform support goals are for Ardi.)

per1234, 2021-09-06:
I think it must be controllable via the application because you can configure this behavior when the Arduino Serial Monitor opens a port via the boards platform configuration:
https://arduino.github.io/arduino-cli/latest/platform-specification/#serial-monitor-control-signal-configuration

egnor, 2021-09-06:
(kinda weird that's at the board platform configuration level, seems more like an end user preference...)

per1234, 2021-09-07:
Yeah, but the Arduino development software is intended to provide a gentle learning curve. In this interest, it's important to make the UI as simple as possible, which means leaving out options that aren't necessary and aren't useful to the majority of users. This control signal configuration is firmly in that category.

Although it's essential that Arduino be beginner friendly, it's also important for it to allow advanced users to do advanced things. If for some reason it is desirable to make this setting easy for the user to change, it can be made a[ custom board option](https://arduino.github.io/arduino-cli/latest/platform-specification/#custom-board-options), defined in boards.txt.

closed 2023-01-02 by robgonnella